CarlosGS/grblForCyclone #3: Servo support

Niekvdplas, 2017-03-23:
Hi Carlos,

I am using your firmware for my school project, for our Z-axis we want to use a 9g servo to move the pen up and down. Is there already servo support in this version which i dont know of? Or will you be implementing this? I hope you might be able to help us or atleast guide us on how to implement this since we are eager to learn

Yours sincerely,,

Niek van der Plas

CarlosGS, 2017-03-28:
Hi Niek! I have not used servos with GRBL.
Maybe you could incorporate the servomotor code in the ramps.h file. There is a rampsInitSteppers() and rampsWriteDisable() function where the servo could be initiated, attached or detached, instead of the Z motor pins.
Also in that file there is rampsWriteSteps() and rampsWriteDirections(), the code to move a servo could potentially go there.

Niekvdplas, 2017-03-28:
Hey Carlos,

I will try what you just said, I hope you know what to do when i don't :D
Thank you very much for your time

Sincerely,

Niek

Niekvdplas, 2017-03-28:
I need to do a few things:

- Make the servo connect to the ramps
- give it a gcode command for it to move

2017-03-28 18:43 GMT+02:00 niek plas <niekvanderplas1@gmail.com>:

SpaceNoodles01, 2017-08-27:
Hi Niekvdplas
Did you archieve something?

praful15, 2018-04-06:
@Niekvdplas i was looking for the same. Please tell us if you have found a solution to that?