CarlosGS/grblForCyclone #4: RAMPS 1.4 Extruders / dual axis motors

raenji-sk, 2017-04-12:
Hi, is it possible to configure the RAMPS 1.4 with your firmware to duplicate the output of motor X to E0 and motor Y to E1? 
Just like on Marlin / Repetier firmware - `#define Y_DUAL_STEPPER_DRIVERS`

CarlosGS, 2017-04-24:
Hi, this is not implemented, but it should be easy to do so:
- First add the definitions for Y2_STEP_PIN, Y2_ENABLE_PIN, Y2_DIR_PIN in https://github.com/CarlosGS/grblForCyclone/blob/grblForCyclone/cpu_map.h#L372
- Then you will have to modify each function in [ramps.h](https://github.com/CarlosGS/grblForCyclone/blob/grblForCyclone/ramps.h#L70), to duplicate each call that affects the Y axis.
Cheers,
Carlos

creisey, 2017-11-15:
could you make the entire example please, im really not used to write definitions.

i would love to have a cloned y axis.

also how to modify the ramps.h

thanks

CarlosGS, 2017-11-15:
Hi, sorry I don't have the time :(
Cloning an axis can also be implemented by physically connecting both motors to the same driver, or even by wiring two drivers in parallel (each driving a different motor).
Cheers!

creisey, 2017-11-15:
thx, i'll try that