CarlosGS/grblForCyclone #5: 400 steps Stepper Motor support?

bmanos, 2017-10-08:
Hello,
I would like to use 400 steps stepper motors (because I have them) 

Specifications:
Step Angle (degrees) : 0.9
2-Phase (4 wires)
Rated Voltage: 3V
Rated Current: 1.7A/Phase
5mm Diameter Drive Shaft
Holding Torque: 48N.cm - 4.8kg.cm
NEMA 17 form factor
datasheet link: http://dlnmh9ip6v2uc.cloudfront.net/datasheets/Robotics/42BYGHM809.PDF

How can I configure them to work?

Thanks in advance
Emmanuel

CarlosGS, 2017-10-08:
It depends on the version of GRBL that you are using, and whether it is a RAMPS of a Sanguinololu board.
Check out the GRBL manual to see if it helps: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#100-101-and-102--xyz-stepsmm

bmanos, 2017-10-08:
Thanks for the quick response
I use the A4988 driver and I also use the your file https://github.com/carlosgs/grblForCyclone/archive/grblForCyclone.zip
Should I modify the "defaults.h"? and what should I change? a little help is very welcome
Thanks

CarlosGS, 2017-10-08:
Yes, I think you should change STEPS_PER_REV almost at the bottom of the defaults.h

Then follow these instructions: https://github.com/CarlosGS/grblForCyclone#how-to-flash-the-grbl-firmware

Let me know of the outcome!

bmanos, 2017-10-09:
Thanks I already did this without success but I clear the eeprom first and now it works. The only problem now is the speed it moves quicker than before (how can I change this)
Thanks one more time

CarlosGS, 2017-10-09:
Please first check that the performed distances are accurate.

bmanos, 2017-10-09:
The distance for Y is ok :101 and for X is 161 instead of 168.
I also correct those values inside the defaults.h file.

CarlosGS, 2017-10-09:
Then it looks fine. The motion speed can be adjusted in the host software. Are you using bCNC?

bmanos, 2017-10-09:
yes

CarlosGS, 2017-10-10:
Awesome, bCNC is the option that I prefer. Have you seen this tutorial? https://github.com/CarlosGS/grblForCyclone/wiki/1)-Configure-bCNC-for-Cyclone

CarlosGS, 2017-10-10:
Sorry this is the right tutorial page: https://github.com/carlosgs/grblForCyclone/wiki/2%29-Hello-World!-Mill-a-bowl-with-bCNC

bmanos, 2017-10-10:
Hi Carlos,
yes I have seen it and I will try it next week end.
Where can I found infos on adding z end stop and for the probes usage.
Is there any other tutorial from start to finish on how to design export and mill a pcb board?
Thanks again, Emmanuel

CarlosGS, 2017-10-11:
Not that I can think of. Give a try to the search option of the Cyclone mail list: https://groups.google.com/group/cyclone-pcb-factory
I have not used the machine for a long time so there are some things I don't remember too well..

And good luck with the tests! Feel free to share your results in the mail list as there is more people that can help.

closed 2017-10-11 by CarlosGS