CarlosGS/grblForCyclone #6: Endstops and probe not functioning

bleachworthy, 2018-06-03:
When homing is enabled and it's given a new strategy (z then x+y for example) the axis crashes into the switch at high speed. If I invert endstop pins in EEPROM it does the same but at a much reduced rate. I have referenced the schematic for the sanguino board and confirmed my pins are correctly set and even tested all the way to the pin on the 1284p to ensure limit switches are indeed working. When I send "?" over serial it does not report any endstop triggering, just current coordinates. I do notice zmin is disabled and replaced with z probe, is that the source of my problem? 

If this wasn't meant to be homed how do you zero it?

CarlosGS, 2018-06-13:
Hi! Indeed the Z axis, with this setup, is not intended to be homed automatically. Instead, it has to be zeroed to the surface of the material. This can be achieved manually "stepping" the Z axis until the tip of the tool barely touches the material, or for conductive materials, using the "probe" command and connecting the Z endstop as an electrode probe.
Cheers!

CarlosGS, 2018-06-13:
Please check out the project's forum where it will be quicker to solve these questions: https://groups.google.com/forum/#!forum/cyclone-pcb-factory