CarlosGS/grblForCyclone #8: LCD I2c not working on mega with ramps 1.4 shield.

rigelinorion, 2018-12-10:
Hello. I am trying to connect a 20x4 lcd display through an i2c convector on a PCF8574 chip. On the example of the LiquidCristal_I2C library everything works without a problem, so the connection is correct. After loading grblForCyclone, the controller is blocked after calling the lcd.init () procedure. Do you have any idea what could cause a conflict? After checking in the lcd library I have the impression that this delay affects the operation of the program. I would not like to use another microcontroller.

CarlosGS, 2018-12-11:
Hi, I have never used a display to control the CNC, I don't know why it could block when initializing the LCD.
Sorry I cannot help!

rigelinorion, 2018-12-11:
Hello. Thanks for the response. The situation has changed since yesterday. The LCD has started. It helped calling lcd_init just before calling protocol_main_loop in the main function. By the way, I have one more question. Is it possible to control the engine using gcode commands from the firmware level?

rigelinorion, 2018-12-12:
I've already got it. It works. I close the thread.

closed 2018-12-12 by rigelinorion