CarlosGS/grblForCyclone #9: Unable to flash my ramps board

talljon, 2019-03-18:
Hi there.  First, please understand I am not a programmer.  I am an airline pilot who tinkers in this, so excuse my nativity.  

I am trying to re-flash my ramps 1.4 board and use it for a homemade milling cnc machine.  I have cleared the EEPROM memory as instructed.  However, when I try to flash my board with the GRBL firmware I receive the error message below while it is compiling.  In my limited knowledge it looks like I need to point/include the other files in the zip folder which contained the .ino file that I opened in the IDE.  However, I don't understand how to do this simple task (if this is my problem).  I have tried to research this and learn on my own and now I'm humbly asking for some help.

Again, I apologize that I'm not understanding this.

Thanks for your help,

Jon

Arduino: 1.6.11 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Tyrion\Downloads\grblForCyclone-grblForCyclone\grblForCyclone-grblForCyclone\grblForCyclone\grblForCyclone.ino:35:20: fatal error: system.h: No such file or directory

 #include "system.h"

                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
.

CarlosGS, 2019-03-18:
Hi! I believe this is due to the directory "grblForCyclone-grblForCyclone"
being created by Arduino. You see, it forces you to have the same name for
the .ino as the folder containing it.
Give a try to renaming the main folder "grblForCyclone" containing all the
.c, .h and the .ino, see if it works

talljon, 2019-03-18:
Carlos,

That is exactly what the problem was, I corrected like you suggested and it flashed perfectly.  Thank you very much!  I wish I understood more about this, but I love learning and people like you are helping me learn more and more everyday.

If you ever have questions about flying airplanes, maybe I could help you out.

Thank you again,

Jon

CarlosGS, 2019-03-18:
No problem! Good luck and happy milling!!

PS: Super interesting that you're a pilot with this side hobby!! I take
your word, though I've only flown drones so far :-)

Cheers!

closed 2019-03-19 by talljon